analogdevicesinc/scopy #824: Oscilloscope: Trace averaging

amiclaus, 2020-07-15:
Add trace averaging feature in the oscilloscope, being able to manually configure the number of sweeps to average.

This feature is available in the ALICE app for M2K:
https://wiki.analog.com/university/tools/m2k/scopy/oscilloscope#using_built-in_signal_measurement_features

This request comes after receiving feedback on StudentZone(Analog Dialogue) - EngineerZone.

amiclaus, 2020-07-15:
Before applying trace avg:
[image]

After applying trace avg:
[image]

damercer, 2020-07-15:
The amplitude went down because you don't have the triggering on. Each waveform buffer added to the average needs to be aligned in time to the same trigger point. 

I don't have a working installation of libm2k bindings in Python right now or I'd show you what actually happens to the noise level when using trace averaging. It is most noticeable in the 1V/div and greater settings where the ADC lsb is 10X larger.

ALICE for M1k uses the same technique, obviously, so I'll show the effect with that. Without trace averaging:

[image]

With trace averaging (8 traces):

[image]

Doug

amiclaus, 2020-07-15:
Thanks for the explanation! By the way, the executable release for ALICE M2K from wiki does not contain the option to set the number of averages.  Is a default value used? I'm just curious.