analogdevicesinc/scopy #842: <Instrument>: <None of the applications for Scopy came in the app when I installed it.>

sean600, 2020-08-05:
OS Version: Windows 7 SP 1 (6.1)
FW Version: 
SW Version: 3d4c582

Description of the bug:<description>
Steps to reproduce:
-
-

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/deuce/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.

adisuciu, 2020-08-05:
Can you be more specific ?

sean600, 2020-08-05:
The circled region: I don’t have any of these parameters when I install the
program.

adisuciu, 2020-08-05:
If you attached an image, I can't see it. 
Did you install the driver and connect to the board in Scopy ?

-Adrian

sean600, 2020-08-05:
Yes I did.

adisuciu, 2020-08-05:
Can you reattach it using the Github web interface. I'm not sure how replying by e-mail works with attachments.
When you select your device, the instruments should appear on the lefthand side. And when you connect they will be enabled. Picture attached

[image]

-Adrian

sean600, 2020-08-05:
> Can you reattach the picture using the Github web interface. I'm not sure

How can I reach you on github?

sean600, 2020-08-05:
[image]
Here's my issue

sean600, 2020-08-05:
I do not have any of the parameters in the circle

adisuciu, 2020-08-10:
I need a screenshot of the actual problem. This is a modified screenshot from the wiki.
The instruments don't appear on the left hand side usually when you haven't connected to the board yet.

Once you connect, they will appear. When I say connect I mean this https://wiki.analog.com/university/tools/m2k/scopy#connecting_to_a_usb_device

-Adrian

closed 2020-10-13 by adisuciu