analogdevicesinc/scopy #863: GR_MODULE no longer available since release of gnuradio 3.8.0.0

RuijieYu, 2020-09-05:
**Environment:**
 - OS: Arch Linux
 - SW Version v1.1.2

**Describe the bug**
Cannot build. CMake cannot find function GR_MODULE. 

**To Reproduce**
Steps to reproduce the behavior:
- install dependencies (cmake, make, python3, git, qt5, gnuradio, mpir, boost)
- then attempt to create CMake build directory

**Expected behavior**
Either build successfully, or fail to *find* a module. 

**Additional context**
Related error messages produced by CMake regarding this issue:
```txt
-- Found VOLK: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/libpython3.so (found suitable exact version "3.8.5") 
CMake Error at CMakeLists.txt:117 (GR_MODULE):
  Unknown CMake command "GR_MODULE".
```

I tried some `grep`'ing around, and noticed that the latest releast of gnuradio that includes GR_MODULE is 3.7.14.0, the release right before 3.8.

adisuciu, 2020-09-07:
Are you trying to build v1.1.2 or latest master ?

v1.1.2 builds with GNURadio 3.7, while latest master builds with GNURadio 3.8. Some other dependencies have changed as well between the two versions.
-Adrian

RuijieYu, 2020-09-07:
Yes I was trying to build v1.1.2, not master. 
Your reply does clean up the confusion, but I had not seen indications of dependencies in the code.. It would help if the dependencies are made clearer.

adisuciu, 2020-10-13:
https://wiki.analog.com/university/tools/m2k/scopy/build-linux 

You can look through older versions of this page for older version build guides.
-Adrian

closed 2020-10-13 by adisuciu