analogdevicesinc/scopy #872: Power Supply: "Enable" did not change to "Disable" after enabling channel

tagoylo, 2020-09-22:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7c68922a 

**Describe the bug**
When enabling the power supply outputs by clicking the "Enable" button, the label "Enable" does not change to "Disable".

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the Power Supply instrument.
2. Click on "Enable" button on both outputs.
4. See the error that the "Enable" label is still there and did not change to "Disable" 

**Expected behavior**
"Enable" should change to "Disable".

**Screenshots**
[image]

AlexandraTrifan, 2020-09-23:
Fixed with #874

tagoylo, 2020-10-20:
This bug is fixed. Closing this item.

PR #873 (2020-09-22): Fix dmm
Should fix: #872
closed 2020-10-20 by tagoylo