analogdevicesinc/scopy #875: Pattern Generator: Displayed waveform is cut

tagoylo, 2020-09-23:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7c68922a

**Describe the bug**

Displayed waveform is cut when the following patterns are selected:

- Clock pattern:
-duty cycle set is multiples of 5

- Random/Binary counter/Pulse/Gray Counter pattern:
-any frequency set results to the displayed waveform cut

**To Reproduce**
Steps to reproduce the behavior:
1. Go to pattern generator
2. Enable a channel, ex. Dio 0
3. Then double click on Dio 0 on plot.
4. Then select any of the patterns: ClockRandom/Binary counter/Pulse/Gray Counter
     -for the clock pattern set duty cycle to multiples of 5 to see the issue.

**Expected behavior**
Waveform should not be cut.

**Screenshots**
[image]
[image]
[image]

DanielGuramulta, 2020-09-28:
Fixed with: c6e4092ca692b73701978548d3b71301940b7e61

tagoylo, 2020-10-21:
This bug is fixed. Closing this item.

tagoylo, 2021-04-07:
Issue reappeared on Scopy v1.3 build 441fd19.

tagoylo, 2021-04-28:
Issue still exists in v1.2 RC2.

tagoylo, 2022-03-25:
Issue still exists in v1.4.0 rc1.

adisuciu, 2022-03-29:
https://github.com/analogdevicesinc/scopy/pull/1264

tagoylo, 2022-04-27:
Bug verified fixed on v1.4.0 rc2. Closing this item.

closed 2022-04-27 by tagoylo