analogdevicesinc/scopy #876: Pattern Generator: Run and Single buttons messy sequence

tagoylo, 2020-09-23:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7c68922

**Describe the bug**
Clicking Run does not change the label to Stop.  
Although with the following sequence, the Stop label shows.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Pattern generator
2. Click Run 
3. Then click Single.
4. The Stop label shows with green box.
5. Then if we click Stop (with green box), it now changes to orange box and the Single will also change to Stop label.
6. Now both buttons are with Stop labels and to get back to their original state, click 3 times
on the supposed-to-be Single button with Stop label.

**Expected behavior**
Clicking Run should change the label to Stop. And by clicking Single, Stop should revert back to Run.

**Screenshots**
[image]

DanielGuramulta, 2020-09-28:
Fixed with: 625b2e0c40563ff6d70e9c628da2c6db971ad6c3

tagoylo, 2020-10-20:
This bug is fixed. Closing this item.

closed 2020-10-20 by tagoylo