analogdevicesinc/scopy #877: Pattern Generator: Clicking Single button has no response

tagoylo, 2020-09-23:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7c68922

**Describe the bug**
When clicking Single, there is no response in the instrument.

**To Reproduce**
Steps to reproduce the behavior:
1. Open pattern generator
2. Enable a channel, Dio 0
3. Then double click on Dio 0 on the plot and change pattern to clock
4. Then click Single
5. There is no response on the instrument

**Expected behavior**
When clicking Single, the color of the button should momentarily in darker orange and the white box beside the instrument name should momentarily turn to a green triangle indicating it ran for a moment.

DanielGuramulta, 2020-09-28:
Fixed with: 0a601107ef19f5f8e69f77efdf2601c0996f3e7f

tagoylo, 2020-10-20:
This bug is fixed. Closing this item.

closed 2020-10-20 by tagoylo