analogdevicesinc/scopy #878: Digital IO: Automatically set group value to max value if value entered is above max value

tagoylo, 2020-09-24:
**Is your feature request related to a problem? Please describe.**
Taking into account new users, when setting the value of group DIO channels, when the value entered is above 255 (which is the max value), it has no response. 
To reproduce:
1. Go to Digital IO
2. Set channels to Group. Then set Out.
3. Type in value greater than 255, ex 258
4. There is no response.

New users may not know yet the max value.

**Describe the solution you'd like**
Automatically set group value to max value if value entered is above max value.
Just like in power supply when entering values greater than 5V, the value is automatically set to 5V.

adisuciu, 2020-10-09:
https://github.com/analogdevicesinc/scopy/commit/3f09986fc7ef5f404df08aa8e64e5659138533f2 fixed

tagoylo, 2020-10-20:
Enhancement is implemented. Closing this item.

closed 2020-10-20 by tagoylo