analogdevicesinc/scopy #880: Pattern Generator: UART decoder does not display correct data

tagoylo, 2020-09-24:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d6

**Describe the bug**
UART decoder does not display correct data on the plot.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to pattern generator
2. Enable a channel, ex DIO 0
3. Double click on it on the plot to open its channel settings.
4. Set pattern to UART
5. Set data, for example 1111
6. It does not show the data set on the plot. 

**Screenshots**
[image]

DanielGuramulta, 2020-09-28:
Fixed with: 9dc1debcc7ff56302d8002ee9b5647a84d0a1087

tagoylo, 2020-10-21:
This bug is fixed. Closing this item.

closed 2020-10-21 by tagoylo