analogdevicesinc/scopy #881: Pattern Generator: Scopy crashes when changing clock frequency (fine interval enabled)

tagoylo, 2020-09-24:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d

**Describe the bug**
Scopy crashes when changing the frequency down using the (-) button and fine interval is enabled (knob is with orange dot). This happens when at least two channels are enabled and pattern is selected.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable at least 2 Dio channels.
2. Select Clock pattern for both channels.
3. Go to first channel and enable fine interval for the clock frequency.
4. Then try increasing and then decreasing the frequency using the +/- buttons.
5. It is observed that after a few clicks on (-), Scopy lags and then crashes.

**Expected behavior**
Scopy should not crash.

**Screenshots**

[image]

**Additional context**

[ini.bak and dmp files.zip](https://github.com/analogdevicesinc/scopy/files/5274536/ini.bak.and.dmp.files.zip)

DanielGuramulta, 2020-09-28:
Fixed with: d77bc7525dd04a060d6be9861f233c43c2ad1117

tagoylo, 2020-10-21:
This bug is fixed. Closing this item.

closed 2020-10-21 by tagoylo