analogdevicesinc/scopy #882: Pattern Generator: In number pattern, the color of the number set is always red after entering

tagoylo, 2020-09-24:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d

**Describe the bug**
 In number pattern, the color of the number set is always red after entering a value although it is within the limit.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the pattern generator
2. Enable a channel, ex. Dio 0
3. Then set its pattern to Number. By default, the number is 0 and the color is not in red.
4. Type 0 again on the number box, now the color of 0 is red.

**Expected behavior**
Only red when the number is above the max number of grouped channels' capacity.

**Screenshots**
[image]

DanielGuramulta, 2020-09-28:
Fixed with: 8fb53d4c46163a20faf9a831c31d522a48583efa

tagoylo, 2020-10-20:
This bug is fixed. Closing this item.

closed 2020-10-20 by tagoylo