analogdevicesinc/scopy #884: Pattern Generator: In binary counter pattern, the parallel items do not correspond to the timing diagram/plot

tagoylo, 2020-09-25:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d

**Describe the bug**
In binary counter pattern, the parallel items do not correspond to the timing diagram/plot. It also does not correspond to the number of channels grouped.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to pattern generator
2. Enable at least 2 channels, Dio 0 and Dio 1
3. Then group the 2 channels.
4. Select group pattern as binary counter
5. Observe how the parallel items plot only has 0 displayed instead of having 0, 1, 2, 3. It does not correspond. It's always like a bit/s behind.
See screenshots for reference.

**Screenshots**
[image]
[image]

DanielGuramulta, 2020-09-28:
Fixed with: afb62be43fd86badff4f998aa596433cea0eb14e

tagoylo, 2020-10-21:
This bug is fixed. Closing this item.

closed 2020-10-21 by tagoylo