analogdevicesinc/scopy #887: Preferences: Clearer statement to where the user should find the notes

HRosete, 2020-09-25:
**Environment:**
OS: Windows 10
FW Version: m2k-fw-v0.25-3-g1c38-dirty
SW Git Tag: 7eec50d6

**Is your feature request related to a problem? Please describe.**
Enabling the user notes in the Preferences says that it can be found on the main page. A new user will be confused about where the "main page" is. One may assume that it is in the Home Page:
[image]
when in fact, it is located in a tab under the Preferences.
[image]

**Describe the solution you'd like**
A better reword of the "Enable user notes in main page," befitting the right location of the user notes tab to avoid confusion.

HRosete, 2020-10-06:
FW: 
[m2k-fw-v0.25-3-g1c38-dirty.zip](https://github.com/analogdevicesinc/scopy/files/5331268/m2k-fw-v0.25-3-g1c38-dirty.zip)

adisuciu, 2020-10-20:
When the user notes are set, by the user, they are displayed in the main page - as you described it. I think the wording is fine.

closed 2020-10-20 by adisuciu