analogdevicesinc/scopy #889: General Settings (Preferences): Changing the language doesn't translate everything

HRosete, 2020-09-25:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50da

**Describe the bug**
Some parts of the Preference section does not translate to the language of preference.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Preferences"
2. Select language
3. Restart Scopy

**Expected behavior**
Everything in the Preference section must be translated to the selected language

**Screenshots**
scopy_cn:
[image]
scopy_fr:
[image]
scopy_ro:
[image]

adisuciu, 2020-10-23:
https://github.com/analogdevicesinc/scopy/pull/937 working on it

adisuciu, 2020-11-02:
Fixed

closed 2020-11-02 by adisuciu