analogdevicesinc/scopy #890: General Settings (Preferences): No "restart" warning when changing the language from CN

HRosete, 2020-09-25:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50da

**Describe the bug**
Scopy set in scopy_cn (Chinese) language lacks the "restart" warning. It only happens the first time you opt to change the language from Chinese. Succeeding attempts display the warning.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Preferences"
2. Set your language to "scopy_cn"
3. Restart Scopy.
4. Go to "Preferences"
5. Change language back to auto (or scopy_ro, scopy_fr, scopy_en)

**Expected behavior**
An attempt in changing the language must display a Scopy restart warning on the first try.
[image]
(taken on the 2nd attempt to change the language from Chinese)

**Screenshots**
1. 
[image]
2. 
[image]
[image]
[image]

HRosete, 2020-10-06:
FW: [m2k-fw-v0.25-3-g1c38-dirty.zip](https://github.com/analogdevicesinc/scopy/files/5331266/m2k-fw-v0.25-3-g1c38-dirty.zip)

adisuciu, 2020-10-23:
https://github.com/analogdevicesinc/scopy/pull/937 - working on it

adisuciu, 2020-10-23:
https://github.com/analogdevicesinc/scopy/pull/939

closed 2020-10-30 by adisuciu