analogdevicesinc/scopy #891: Network Analyzer: Wrong plot and buffer shows no signal

tagoylo, 2020-09-28:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d

**Describe the bug**
When trying to get the bode plot of a LPF, the bide plot acquired by the network analyzer is not of a LPF. When monitoring the plotting on the buffer previewer, it can be observed that there is no signal shown on the buffer previewer. See the picture attached.

**To Reproduce**
Steps to reproduce the behavior:
1. Set up the following for a LPF plot. 
[image]
2. Set frequency sweep from 10Hz to 1MHz.
3. Set sample count to 200.
4. Set amplitude to 1V, no offset
5. Enable buffer previewer to monitor buffer.
6. Run network analyzer. Observe that there is no signal showing in the buffer previewer and the plot for LPF is wrong.

**Screenshots**
[image]
[image]

amiclaus, 2020-10-06:
@tagoylo the circuit setup looks like a HPF rather than LPF.

@adisuciu Anyway, this looks like a valid issue:

Same setup as above (same firmware, Scopy version)

Output:
[image]

tagoylo, 2020-10-07:
Issue invalid. Did not show up after checking again. Closing this item.

closed 2020-10-07 by tagoylo