analogdevicesinc/scopy #894: General Settings (Save/Load): Saved profile with 2 groups in pattern generator, loads only the 2 groups in 1 group

tagoylo, 2020-09-29:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d

**Describe the bug**
When saving a profile with 2 groups in pattern generator and then loading that saved profile, loads the 2 groups combined to 1 group instead of loading 2 separate groups. See the screenshot for reference. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the pattern generator
2. Create two groups of channels.
3. Save profile as profile1.ini
4. Close scopy and reconnect.
5. Load profile1.ini
6. Observe that instead of 2 separate groups, the loaded settings in the pattern generator is only 1 group which basically the 2 groups combined.

**Screenshots**
[image]

DanielGuramulta, 2020-10-02:
Should be fixed with: 0057a822320e1389390eb5a25c93ce4411f36a82

tagoylo, 2020-10-21:
The groups were loaded separately. But when I selected a channel first then loaded the profile, the selected channel is added to the first group. For this, I save a profile with 2 groups, Dio0-Dio1 and Dio2-Dio3.  Then, reset scopy then enable other channels in pattern generator. Select a channel, then load the profile. 
[image]

DanielGuramulta, 2020-10-30:
Should be fixed with #947

closed 2021-05-05 by adisuciu