analogdevicesinc/scopy #896: Logic Analyzer: External trigger condition Low trigger acted as High trigger and vice versa

tagoylo, 2020-09-29:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d

**Describe the bug**
When using the external trigger condition Low Trigger, then trying to run the logic analyzer and triggering for the plot, the logic analyzer initiated capture when the trigger is set to high instead of low.
This also happens when the condition in the external trigger is set to high trigger, it only responds to low instead of the high trigger input.
It seems the functionality of Low trigger and High trigger are swapped.

**To Reproduce**
Steps to reproduce the behavior:
Connect DIO0 to T1.
1. From Digital I/O enable DIO 0 as output and set it initially as 0.
2. From Pattern generator enable DIO1 or any other channel and generate a signal.
3. In Logic analyzer Trigger settings enable External Trigger and set low trigger condition.
4. Run logic analyzer, logic analyzer should start capturing signals but it is waiting.
5. From Digital I/O, toggle DIO0 to 1. The logic analyzer will start capture.

**Screenshots**

[image]

DanielGuramulta, 2020-09-30:
Fixed with: 5c40cbb7b34e97912e5765c9d18c23f0812ce157

tagoylo, 2020-10-21:
This bug is fixed. Closing this item.

closed 2020-10-21 by tagoylo