analogdevicesinc/scopy #897: Logic Analyzer: Parallel decoder does not display hex format of number

tagoylo, 2020-09-29:
**Environment:**
 - OS: Windows 10
 - FW Version:  m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d

**Describe the bug**
When sending a number pattern from the pattern generator and checking it on logic analyzer, there is no data in hex format displayed in the parallel decoder.  The data lines are correct but no data is shown in the parallel decoder.

**To Reproduce**
Steps to reproduce the behavior:
1. Set a group in pattern generator with number as pattern (_also there is no parallel decoder with hex data that automatically shows unlike with binary counter_)
2. Add a parallel decoder in logic analyzer.
3. Set up parallel decoder data lines. Monitor the group in logic analyzer. 
4. Run pattern generator and start capture on logic analyzer.
5. It can be observe that although data lines are correct, there is no data is shown in parallel decoder

**Screenshots**
[image]

DanielGuramulta, 2020-09-30:
The parallel decoder will need a clock line in this case. When no clock is specified, the decoder: `"... works in asynchronous mode and samples the data lines upon each transition on any of the data lines."` In the case of a number pattern, there are no transitions.
https://sigrok.org/wiki/Protocol_decoder:Parallel

tagoylo, 2020-09-30:
Thank you for the clarification. Closing this item.

closed 2020-09-30 by tagoylo