analogdevicesinc/scopy #899: Spectrum Analyzer: Axis scale lines messed up after changing unit from V/√Hz to dBFS/dBu/dBV

tagoylo, 2020-09-30:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d

**Describe the bug**
When changing units, from V/√Hz to dBFS/dBu/dBV the graticule axis scale lines is messed up.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the spectrum analyzer
2. Go to sweep settings and change the unit to V/√Hz
3. Then change back to dBFS/dBu/dBV
4. Then observe how the axis lines are compressed at the bottom part of the plot.

**Screenshots**
[image]

AlexandraTrifan, 2020-10-12:
This should be fixed after commit bc38e73db07697b042f474819f6e3adf4275417b

tagoylo, 2020-10-20:
This bug is fixed. Closing this item.

closed 2020-10-20 by tagoylo