analogdevicesinc/scopy #900: Spectrum Analyzer: Importing saved Ref channel shows no name when selecting channels

tagoylo, 2020-09-30:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d

**Describe the bug**
Importing saved Ref channel shows no name when selecting channels. See screenshot for reference.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a Ref channel by using Snapshot.
2. Then export Ref channel using Export in the Ref channel settings.
3. Then import reference channel by clicking + beside channel 2.
4. Browse the saved file.
5. Then click on Channels dropdown menu to see available channels. It can be observed that there is no name shown.

I also attached the Ref csv file I've used.

**Screenshots**
[image]

**Additional context**

[ref.zip](https://github.com/analogdevicesinc/scopy/files/5303667/ref.zip)

AlexandraTrifan, 2020-10-12:
This should be fixed after commit 6e2c6ae35c3722a67864cc0bbae630ea9b9594aa

tagoylo, 2020-10-20:
Importing REF channel now shows CH on the list. Issue fixed. Closing this item.

closed 2020-10-20 by tagoylo