analogdevicesinc/scopy #901: Signal Generator: Skewed Square Waveform from 5MHz up tp 30MHz

HRosete, 2020-09-30:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d

**Describe the bug**
The preview of the square waveform skews into somewhat a trapezoidal waveform from frequencies 5MHz up to 30MHz.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Signal Generator.'
2. Click on 'Waveform' and select 'Square.'
3. Set the frequency anywhere between 5MHz and 30MHz.
4. Increment/Decrement the frequency using the + and - keys and see the waveform skew.

**Expected behavior**
The square waveform must retain its shape even if the frequency's changed.

**Screenshots**
[image]

**Additional context**
(can't seem to upload the .ini file)

adisuciu, 2020-09-30:
This is the expected behavior. Plot sample rate is 75MHz (the same as DAC sample rate)

closed 2020-09-30 by adisuciu

HRosete, 2020-10-01:
Thank you for the clarification, Adrian!