analogdevicesinc/scopy #902: Signal Generator: Visible Falling Ramp Waveform's long overshoot in the preview

HRosete, 2020-09-30:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d

**Describe the bug**
Unnecessary overshoots are seen in the preview of the Falling Ramp Sawtooth's waveform. This is visible on both Channels 1 and 2. This does not appear on the oscilloscope, though.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Signal Generator.'
2. Click on 'Waveform.' Select Falling Ramp Sawtooth.
3. Set the amplitude to 5Vp-p. See the preview graph.
4. Increment/Decrement the amplitude and observe the waveform.

**Expected behavior**
There should be no overshoots on the waveform previews.

**Screenshots**
Channel 1
[image]
Channel 2
[image]

adisuciu, 2020-10-05:
https://github.com/analogdevicesinc/gr-scopy/commit/f48b45cabe2a81f491137dbc220032ea65f080c3 should fix it

HRosete, 2020-10-21:
Verified on 33ca72f9, bug is fixed. Closing this issue.

closed 2020-10-21 by HRosete