analogdevicesinc/scopy #903: scopy fail to open

paa180, 2020-10-02:
OS Version: Windows 10 (10.0)
FW Version: 
SW Version: 3d4c582

Description of the bug:<description>
Steps to reproduce:
-
-

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/phili/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.

AlexandraTrifan, 2020-10-03:
Hi,

Did you try to install the latest release? You can find it using the following link: ( https://github.com/analogdevicesinc/scopy/releases/tag/v1.2.0-rc.1). Does this also happen on the latest version?

From the git hash sw version it looks like you are trying to use the previous (v1.1.2) version. A lot of bug fixes and features were added in the latest version.

Thank you,
-Alexandra

adisuciu, 2020-10-21:
This should be fixed now. Reopen if you are still having issues.

closed 2020-10-21 by adisuciu