analogdevicesinc/scopy #907: Oscilloscope: Fails to autoset horizontal and vertical settings on math generated waveform from SG

HRosete, 2020-10-05:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d

**Describe the bug**
The oscilloscope fails to autoset the Horizontal (Time Base) and Vertical (Volts/Div) settings when the signal generated is from the Math function of the Signal Generator. The result is a cut waveform on the Oscilloscope's screen when it should have displayed several periods of the waveform upon clicking 'Autoset.'

**To Reproduce**
Steps to reproduce the behavior:
1. Open Scopy and connect the M2K to your laptop.
2. Unzip the attached file 'scopy_sig-gen_math.zip' and load the .ini file to Scopy. 
[scopy_sig-gen_math.zip](https://github.com/analogdevicesinc/scopy/files/5325541/scopy_sig-gen_math.zip)

3. Connect AWG CH1 to scope CH1+ and scope CH1- to GND.
4. Click 'Run' on the Signal Generator.
5. Click 'Run' on the Oscilloscope.
6. Click 'Autoset' and observe the waveform read by the oscilloscope.

**Expected behavior**
The oscilloscope should automatically adjust the horizontal and vertical settings to display several periods of the waveform upon clicking 'Autoset.'

**Screenshots**
Waveform using the Math function of the SG:
[image]

Waveform read by the oscilloscope after clicking 'Autoset':
[image]

Expected behavior; the same waveform after adjusting the Horizontal and Vertical settings manually:
[image]

HRosete, 2020-10-06:
FW: [m2k-fw-v0.25-3-g1c38-dirty.zip](https://github.com/analogdevicesinc/scopy/files/5331220/m2k-fw-v0.25-3-g1c38-dirty.zip)
Math equation on SG: f(t) = sin(2*pi*1000*t)

adisuciu, 2020-10-13:
duplicate of https://github.com/analogdevicesinc/scopy/issues/913

closed 2020-10-13 by adisuciu