analogdevicesinc/scopy #916: Network Analyzer: Nyquist plot stops plotting after first plot when capturing in continuous

tagoylo, 2020-10-07:
**Environment:**
 - OS: [e.g. iOS]
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d

**Describe the bug**
When running network analyzer in continuous capture, the Nyquist plot seems to stop capturing/plotting after plotting the first set of data. Also, there's a line that shows connecting the end and start of the plot.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the Network analyzer.
2. I used an LPF circuit (with 470ohms and 1uF for the screenshot below) and used the following setup.
[image]
3. Run network analyzer.
4. Change the plot to Nyquist and observe the plot.
5. The result is that there's a line that shows connecting the end and start of the plot. And it seems that it stopped plotting the newly captured data, unlike Nichols and bode plot.

I attached the result when running only in Single and when running continuously.

**Screenshots**
[image]

adisuciu, 2020-10-09:
https://github.com/analogdevicesinc/scopy/commit/7a500983a67fcefae4d50d10cee62fefbe2a2ef1 fixed

tagoylo, 2020-10-20:
This bug is fixed. Closing this item.

closed 2020-10-20 by tagoylo