analogdevicesinc/scopy #917: pluto: unable to compile with -Werror=uninitialized (default)

ZeroChaos-, 2020-10-07:
**Environment:**
 - OS: Gentoo Linux
 - FW Version: 0.32
 - git tag 46de233fe86d3dcde6b5d7e1898459a6daa345e5

**Describe the bug**
Unable to build with default cflags

I am trimming the output severely, please let me know if it's too severe
```
[223/254] /usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_ALL_DYN_LINK -DFFTW3F_THREADS -DHAS_CONSTEXPR=1 -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_NO_DEBUG_OUTPUT=1 -DQT_NO_KEYWORDS -DQT_QML_LIB -DQT_SVG_LIB -DQT_UITOOLS_LIB -DQT_WIDGETS_LIB -DQT_XML_LIB -I. -I/var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999 -Iscopy_autogen/include -I/usr/include/qwt6-multiaxes -I/usr/include/python3.8 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/glibmm-2.4 -I/usr/lib64/glibmm-2.4/include -I/usr/lib64/libffi/include -I/usr/include/sigc++-2.0 -I/usr/lib64/sigc++-2.0/include -I/var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src -I/breakpad -isystem /usr/include/qt5 -isystem /usr/include/qt5/QtWidgets -isystem /usr/include/qt5/QtGui -isystem /usr/include/qt5/QtCore -isystem /usr/lib64/qt5/mkspecs/linux-g++ -isystem /usr/include/qt5/QtConcurrent -isystem /usr/include/qt5/QtQml -isystem /usr/include/qt5/QtNetwork -isystem /usr/include/qt5/QtUiTools -isystem /usr/include/qt5/QtSvg -isystem /usr/include/qt5/QtXml  -Os -march=native -mtune=native -pipe -frecord-gcc-switches   -Werror=return-type -Werror=uninitialized -Werror=init-self -Werror=switch -fPIC -std=c++11 -MD -MT CMakeFiles/scopy.dir/src/scroll_filter.cpp.o -MF CMakeFiles/scopy.dir/src/scroll_filter.cpp.o.d -o CMakeFiles/scopy.dir/src/scroll_filter.cpp.o -c /var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/scroll_filter.cpp
[224/254] /usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_ALL_DYN_LINK -DFFTW3F_THREADS -DHAS_CONSTEXPR=1 -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_NO_DEBUG_OUTPUT=1 -DQT_NO_KEYWORDS -DQT_QML_LIB -DQT_SVG_LIB -DQT_UITOOLS_LIB -DQT_WIDGETS_LIB -DQT_XML_LIB -I. -I/var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999 -Iscopy_autogen/include -I/usr/include/qwt6-multiaxes -I/usr/include/python3.8 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/glibmm-2.4 -I/usr/lib64/glibmm-2.4/include -I/usr/lib64/libffi/include -I/usr/include/sigc++-2.0 -I/usr/lib64/sigc++-2.0/include -I/var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src -I/breakpad -isystem /usr/include/qt5 -isystem /usr/include/qt5/QtWidgets -isystem /usr/include/qt5/QtGui -isystem /usr/include/qt5/QtCore -isystem /usr/lib64/qt5/mkspecs/linux-g++ -isystem /usr/include/qt5/QtConcurrent -isystem /usr/include/qt5/QtQml -isystem /usr/include/qt5/QtNetwork -isystem /usr/include/qt5/QtUiTools -isystem /usr/include/qt5/QtSvg -isystem /usr/include/qt5/QtXml  -Os -march=native -mtune=native -pipe -frecord-gcc-switches   -Werror=return-type -Werror=uninitialized -Werror=init-self -Werror=switch -fPIC -std=c++11 -MD -MT CMakeFiles/scopy.dir/src/signal_generator.cpp.o -MF CMakeFiles/scopy.dir/src/signal_generator.cpp.o.d -o CMakeFiles/scopy.dir/src/signal_generator.cpp.o -c /var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/signal_generator.cpp
FAILED: CMakeFiles/scopy.dir/src/signal_generator.cpp.o 
/usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_ALL_DYN_LINK -DFFTW3F_THREADS -DHAS_CONSTEXPR=1 -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_NO_DEBUG_OUTPUT=1 -DQT_NO_KEYWORDS -DQT_QML_LIB -DQT_SVG_LIB -DQT_UITOOLS_LIB -DQT_WIDGETS_LIB -DQT_XML_LIB -I. -I/var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999 -Iscopy_autogen/include -I/usr/include/qwt6-multiaxes -I/usr/include/python3.8 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/glibmm-2.4 -I/usr/lib64/glibmm-2.4/include -I/usr/lib64/libffi/include -I/usr/include/sigc++-2.0 -I/usr/lib64/sigc++-2.0/include -I/var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src -I/breakpad -isystem /usr/include/qt5 -isystem /usr/include/qt5/QtWidgets -isystem /usr/include/qt5/QtGui -isystem /usr/include/qt5/QtCore -isystem /usr/lib64/qt5/mkspecs/linux-g++ -isystem /usr/include/qt5/QtConcurrent -isystem /usr/include/qt5/QtQml -isystem /usr/include/qt5/QtNetwork -isystem /usr/include/qt5/QtUiTools -isystem /usr/include/qt5/QtSvg -isystem /usr/include/qt5/QtXml  -Os -march=native -mtune=native -pipe -frecord-gcc-switches   -Werror=return-type -Werror=uninitialized -Werror=init-self -Werror=switch -fPIC -std=c++11 -MD -MT CMakeFiles/scopy.dir/src/signal_generator.cpp.o -MF CMakeFiles/scopy.dir/src/signal_generator.cpp.o.d -o CMakeFiles/scopy.dir/src/signal_generator.cpp.o -c /var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/signal_generator.cpp
/var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/signal_generator.cpp: In constructor ‘adiscope::SignalGenerator::SignalGenerator(iio_context*, adiscope::Filter*, adiscope::ToolMenuItem*, QJSEngine*, adiscope::ToolLauncher*)’:
/var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/signal_generator.cpp:435:24: warning: ‘QVariant qVariantFromValue(const T&) [with T = unsigned int]’ is deprecated: Use QVariant::fromValue() instead. [-Wdeprecated-declarations]
  435 |     qVariantFromValue(i));
      |                        ^
In file included from /usr/include/qt5/QtCore/qlocale.h:43,
                 from /usr/include/qt5/QtCore/qtextstream.h:46,
                 from /usr/include/qt5/QtCore/qdebug.h:49,
                 from /usr/include/qt5/QtCore/qloggingcategory.h:44,
                 from /usr/include/qt5/QtCore/QLoggingCategory:1,
                 from /var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/logging_categories.h:23,
                 from /var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/signal_generator.cpp:21:
/usr/include/qt5/QtCore/qvariant.h:528:17: note: declared here
  528 | inline QVariant qVariantFromValue(const T &t)
      |                 ^~~~~~~~~~~~~~~~~
/var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/signal_generator.cpp:435:24: warning: ‘QVariant qVariantFromValue(const T&) [with T = unsigned int]’ is deprecated: Use QVariant::fromValue() instead. [-Wdeprecated-declarations]
  435 |     qVariantFromValue(i));
      |                        ^
In file included from /usr/include/qt5/QtCore/qlocale.h:43,
                 from /usr/include/qt5/QtCore/qtextstream.h:46,
                 from /usr/include/qt5/QtCore/qdebug.h:49,
                 from /usr/include/qt5/QtCore/qloggingcategory.h:44,
                 from /usr/include/qt5/QtCore/QLoggingCategory:1,
                 from /var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/logging_categories.h:23,
                 from /var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/signal_generator.cpp:21:
/usr/include/qt5/QtCore/qvariant.h:528:17: note: declared here
  528 | inline QVariant qVariantFromValue(const T &t)
      |                 ^~~~~~~~~~~~~~~~~
/var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/signal_generator.cpp: In member function ‘double adiscope::SignalGenerator::get_best_sample_rate(unsigned int)’:
/var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/signal_generator.cpp:2131:39: warning: ‘template<class T> class qLess’ is deprecated: Use std::less [-Wdeprecated-declarations]
 2131 |   qSort(values.begin(), values.end(), qLess<double>());
      |                                       ^~~~~
In file included from /usr/include/qt5/QtCore/qdebug.h:44,
                 from /usr/include/qt5/QtCore/qloggingcategory.h:44,
                 from /usr/include/qt5/QtCore/QLoggingCategory:1,
                 from /var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/logging_categories.h:23,
                 from /var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/signal_generator.cpp:21:
/usr/include/qt5/QtCore/qalgorithms.h:153:40: note: declared here
  153 | class QT_DEPRECATED_X("Use std::less") qLess
      |                                        ^~~~~
/var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/signal_generator.cpp:2131:54: warning: ‘void qSort(RandomAccessIterator, RandomAccessIterator, LessThan) [with RandomAccessIterator = __gnu_cxx::__normal_iterator<double*, std::vector<double> >; LessThan = qLess<double>]’ is deprecated: Use std::sort [-Wdeprecated-declarations]
 2131 |   qSort(values.begin(), values.end(), qLess<double>());
      |                                                      ^
[... cut ...]
/var/tmp/portage/net-wireless/scopy-9999/work/scopy-9999/src/signal_generator.cpp:1197:12: error: ‘nr_of_samples’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
 1197 |   uint32_t nr_of_samples;
      |            ^~~~~~~~~~~~~
cc1plus: some warnings being treated as errors
ninja: build stopped: subcommand failed.
``

if I remove -Werror=uninitialized from the CMakeLists.txt then it will build, but immediately crashes when connecting to hardware.  I don't know if that's related or not, but I don't think it's fair to report bugs when I have to ignore an error like this so let's fix this bug first :-)

adisuciu, 2020-10-08:
I don't think the error message is related to the crash. For building we're using the gcc version that is provided by ubuntu. However based on the title of the issue and the firmware version you provided, I think you are trying to use Scopy to connect to the ADALM-PLUTO. This is not currently supported. We're currently supporting only the ADALM2000 device with Scopy. 

-Adrian

ZeroChaos-, 2020-10-08:
lol, well that would explain my crash I guess.  I'm super sad to hear that though, I hate to take this bug off topic but.... would a donation of a pluto help add support? :-)  I want to use my pluto as a Network Anaylzer like the windows "SATSAGEN" software does.

adisuciu, 2020-10-08:
Thanks for the offer, but we have enough PLUTOs already. We're working on bringing support for other IIO-based devices to Scopy, however this requires some infrastructure changes in the software. Stay tuned for more updates.

-Adrian

ZeroChaos-, 2020-10-08:
Tuned I shall stay.  As for this bug, I'm using gcc 9.3.0 on gentoo.  If you are on a lower version then this bug is likely coming for you soon.

adisuciu, 2020-10-09:
This is what I'm using.
```
gcc --version
gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I'm not getting the error. You can probably initialize that value to 0 and it will work fine. We will also fix that at some point.

-Adrian

closed 2020-10-12 by adisuciu