analogdevicesinc/scopy #919: Network Analyzer: "Stop" label does not revert back to "Run" after clicking Single

tagoylo, 2020-10-08:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 7eec50d

**Describe the bug**
"Stop" label does not revert back to "Run" after clicking Single. See steps to reproduce.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the network analyzer
2. Click Run.
3. Then after a few seconds, click Single.
4. It can be observed that the Stop label did not revert back to the Run label although the function and color of the button are for Run.
5. The if you click on "Stop" in green, the Single button becomes a Stop button but when clicked it still functions as single capture.

**Screenshots**
[image]
[image]
[image]

DanielGuramulta, 2020-10-09:
This should be fixed with commit: 625b2e0c40563ff6d70e9c628da2c6db971ad6c3

tagoylo, 2020-10-20:
Issue fixed. Closing this item.

closed 2020-10-20 by tagoylo