analogdevicesinc/scopy #931: Pattern Generator: First and last item in Parallel decoder are not displayed

tagoylo, 2020-10-21:
**Environment:**
 - OS: Windows 10
 - FW Version: m2k-fw-v0.25-3-g1c38-dirty
 - SW Version or git tag: 33ca72f

**Describe the bug**
The first and last item in parallel decoder are not displayed. See screenshot.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the Pattern generator
2. Click on create a group and select binary counter as a pattern
3. Notice how the first and last items are not displayed.

**Screenshots**
[image]

[image]

DanielGuramulta, 2020-10-21:
https://sigrok.org/wiki/Protocol_decoder:Parallel
In the case of the pattern generator the decoder works in asynchronous mode and samples upon each transition on each of the specified data lines. There are no transitions at the beginning of the plot nor at the end of it. This introduces the shown gaps on the attached images. This doesn't seem like a bug because the decoder that is used here works as it should.

tagoylo, 2020-10-21:
Thanks for the clarification. Closing this item as it is invalid.

closed 2020-10-21 by tagoylo