analogdevicesinc/scopy #932: General Settings: Hidden Debug Instrument does not appear when invoked with unplugging M2K while Scopy is running

HRosete, 2020-10-21:
**Environment:**
 - OS: Windows 10
 - FW Version: [m2k-fw-v0.25-3-g1c38-dirty.zip](https://github.com/analogdevicesinc/scopy/files/5413412/m2k-fw-v0.25-3-g1c38-dirty.zip)
 - SW Version or git tag - 33ca72f9 

**Describe the bug**
Enabling the debug instrument on the preferences doesn't do anything. I also enabled the debug messages for when issues arise. Antoniu said that enabling the debug instrument invokes a new instrument to appear in the instruments. He said that there is a hidden debugger in Scopy where you can access the onboard devices and registers. I haven't seen it.

**To Reproduce**
Steps to reproduce the behavior:
1. Attach M2K to your laptop using a micro-USB to USB cable.
2. Open Scopy. Make sure to disconnect all connections on M2K.
3. Go to "Preferences." Under 'Debug', enable debug messages and the debug instrument.
[image]
4. Attach CH1+ to W1 and CH1- to GND.
5. Go to Signal Generator and generate a 5Vpp, 1kHz signal (or any signal of choice).
6. Run Signal Generator and Oscilloscope.
7. While the instrument still runs, unplug M2K from your laptop.
8. Wait for the debug messages to appear.
9. Wait for the debug tool to appear in the instrument' pane. Observe.

**Expected behavior**
The hidden debugger instrument should appear in the instruments.

**Screenshots**
[image]

adisuciu, 2020-10-21:
The terminology is a little confusing. Debug instrument is an instrument that you can access registers with - similar to Debug plugin in iio-oscilloscope. You need to restart Scopy after enabling that plugin. I'll add this to the preferences menu.

Debug messages are the messages you received as popup when you disconnected the board. Maybe this should be renamed to Enable DebugMode ?

HRosete, 2020-10-22:
Oh, that must be it. I haven't restarted Scopy after enabling the Debug Instrument. Should try that. Are there any more things I should look for when enabling the debug instrument? Any prerequisites, perhaps?

Maybe, probably. But I think these separate checkboxes for the debug messages and the debug instrument are good. Maybe we should put a notice on the debug instrument that you should restart Scopy to enable this. That way it won't be too confusing.

adisuciu, 2020-10-23:
I added them, and renamed debug instrument to IIO Debug instrument.

closed 2020-10-30 by adisuciu