analogdevicesinc/scopy #934: Oscilloscope: Shifting the phase on Signal Generator does not implement on the Oscilloscope real-time

HRosete, 2020-10-21:
**Environment:**
 - OS: Windows 10
 - FW Version: [m2k-fw-v0.25-3-g1c38-dirty.zip](https://github.com/analogdevicesinc/scopy/files/5414324/m2k-fw-v0.25-3-g1c38-dirty.zip)
 - SW Version or git tag: 33ca72f9 

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Attach M2K to your laptop using a micro-USB to USB cable.
2. Open Scopy. Make sure to disconnect all connections on M2K.
3. Unzip Book1-csv.zip (see below) and save the Book1 CSV file.
4. Go to Signal Generator. Set these settings to Channel 1: Triangle waveform, 4Vpp amplitude, 1MHz frequency, 2V offset, 180 degrees phase.
5. Select Channel 2 and got to Buffer. 
6. Click 'Load File' and navigate to Book1.csv. Click 'Enter.'
7. In Channel 2 still, set these settings to Scaling (below 'Load File'): Amplitude = 5V, SampleRate = = 2.027 Msps, Offset = 0V, Phase = 0 samples.
8. Go to the Oscilloscope.
9. Set time base (Horizontal) to 1us, position to 0s, volts/div to 1V/div, and position to 0V.
10. Run Signal Generator and Oscilloscope.
11. Detach the Signal Generator by dragging the instrument away from the pane.
12. Change Channel 1's phase by clicking "+" and "-" and observe the waveforms on the Oscilloscope.
13. Do the same thing to Channel 2, change the phase. Observe.

**Expected behavior**
The waveform seen on the oscilloscope must reflect the changes applied in real-time from a running signal generator.

**Screenshots**
at Phase shift: 90 degrees; Waveform on Oscilloscope corresponds to the Signal Generator's
[image]

at Phase shift: 180 degrees; Waveforms on Oscilloscope and Signal Generators do not match with regards to Phase
[image]

at Phase shift: 225 degrees; Waveforms on Oscilloscope and Signal Generators do not match with regards to Phase
[image]

**Additional context**
[scopy_osc.zip](https://github.com/analogdevicesinc/scopy/files/5414438/scopy_osc.zip)
[Book1-csv.zip](https://github.com/analogdevicesinc/scopy/files/5414439/Book1-csv.zip)

adisuciu, 2020-10-21:
Well you are setting 1M frequency for the triangle and 2.027 for the second. So the signals will never be in phase because their frequencies are not multiples of one another - they are sliding one on top of the other. If you chose 1M frequency for triangle and same frequency for the stairstep, their phase would match the settings.

-Adrian

HRosete, 2020-10-28:
Thank you for the clarification, Adrian.

HRosete, 2020-10-28:
Closing this issue.

closed 2020-10-28 by HRosete