analogdevicesinc/scopy #940: Manual range setting in Scopy Voltmeter

damercer, 2020-10-26:
When using the Voltmeter tool in Scopy I don't see a clear indication of which input range the M2k is in. Sometimes it looks like it might be in high gain or low gain range based on the amount of noise / size of the voltage steps. Does Scopy do some kind of Auto-ranging in the Voltmeter tool? It would be nice it there was some sort of indicator that changed (color maybe?) depending on which range it is in.

As an enhancement having a manual override so that the user can turn off any auto-ranging and select which range is to be used. There could be an "out-of-range" warning if the voltage is at or beyond the range that is manually set.

Doug

adisuciu, 2020-10-27:
I just checked, the voltmeter in Scopy always uses low-gain. I can change it to have LOW/HIGH/AUTO .. 

-Adrian

damercer, 2020-10-27:
That explains why the readings were so much more noisy when measuring a small differential voltage ( < 500 mV ) compared to the mean value measured in the scope on the 0.5 V/div range.

That would be a good feature to enhance the Voltmeter usability for small DC voltages.

Thanks

Doug

adisuciu, 2020-11-09:
We implemented this here: https://github.com/analogdevicesinc/scopy/pull/942

You can give this a test if you want https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/36108722/job/fhvldk7vygj16han/artifacts

It has high gain/low gain/ auto gain. Auto switches from low to high as soon as it detects an overflow, and from high to low after there have been 10 readings below a threshold.

-Adrian

closed 2021-05-05 by adisuciu