analogdevicesinc/scopy #943: M2k: Cannot "run" from square signal

RuijieYu, 2020-10-28:
**Environment:**
 - OS: Windows 10
 - FW Version M2K, purchased in 2020
 - SW Version or git tag 1.2.0rc1 (7eec50d)

**Describe the bug**
Using M2K, in signal generator, signal type set to "square", application freeze when clicking "run". 
In contrast, first set to "sine", run, then switch to "square", the application runs fine and produces expected output

**To Reproduce**
Steps to reproduce the behavior:
1. Connect M2K, go to signal generator
2. Without clicking "run", set type = square, then p-p = 4V, offset = 2V, freq = 2kHz, phase = 0 deg, duty cycle = 50
3. Click "run"
4. Application freezes

**Expected behavior**
Output is produced in oscillator

**Screenshots**

**Additional context**
Default ini file

RuijieYu, 2020-10-28:
Addition:
If I first set channel 1 to sine (2V, 2kHz, 0V offset), then set channel to square (2V, 2kHz, 0V offset), the program does not freeze. (If relevant, the oscilloscope is running)

adisuciu, 2020-10-30:
I cannot reproduce. Can you give it another try using the latest version ?

adisuciu, 2021-03-25:
Closed due to inactivity

closed 2021-03-25 by adisuciu