analogdevicesinc/scopy #945: libusb: crash

yyusea, 2020-10-29:
**Environment:**
 - OS: [windows7]

scopy current use libusb-1.0.21-2,which lead app crash when start.
libusb-1.0.23-1 can fix this problem.

adisuciu, 2020-10-30:
We didn't change anything to avoid issues for this release. We will consider using latest libusb for the next release.
-Adrian

adisuciu, 2021-02-26:
This is not a release but this version should have a lot of fixes - would you mind giving it a try and see how it works?
This version uses libusb 1.0.24-2 on Windows
Make sure you tick delete the ini files in the setup process, or delete them manually.
https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/37944786/job/g4x6c5ep2csfs08o/artifacts
-Adrian

adisuciu, 2021-04-22:
Closing this due to inactivity - reopen if you encounter any more issues ..

closed 2021-04-22 by adisuciu