analogdevicesinc/scopy #948: Scopy for M2K: Scopy crashes at the end of the connection/calibration process on one of my two X1Carbon laptops.

connorka, 2020-10-30:
OS Version: Windows 10 (10.0)
FW Version: v0.25
SW Version: 3d4c582

Description of the bug: Scopy crashes at the end of the connection/calibration process on one of my two X1Carbon laptops. 
Steps to reproduce: I downloaded a new version of Scopy and it still happens. The M2K works fine on my other X1 Carbon, which is a few generations older. 
-
-

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/conno/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.

adisuciu, 2020-10-30:
Hmmm, that's interesting. There are a few things that you could try. Can you change the USB cable and/or USB port. Make sure you connect the M2K to one of those high power USB port. The port should be marked with a small lightning bolt. Alternatively you could use the alternate power adapter for powering the M2K and use the middle port for connecting it to the PC. This should rule out any power issues.

If your problem still persists, can you go to Settings and click the "Enable the Debug Messages" checkbox like so:

[image]

Reproduce the crash.
If any dialog boxes show up, please screenshot them and attach them here.
After the crash, go to C:/Users/conno/AppData/Roaming/ADI/ and zip all the data there - including the crashdmp folder(hopefully it's not empty) and send it over to us.

-Adrian

DanielGuramulta, 2020-10-30:
From the SW version provided you are using an older version of Scopy (v1.1.2).You  can try the latest release from here and see if the issue persists: https://github.com/analogdevicesinc/scopy/releases

-Daniel

connorka, 2020-10-30:
Adrian

I used the same cable with the other laptop so that should not be a problem. I also just tried it again with the external power source and it does the same thing. Everything has worked fine on the newer X1 Carbon for a while. I did a lot of videos on using the M2K over last weekend, which also required running a document camera and Zoom. I did run into one problem when I tried to source a bit too much current in one experiment (from the signal generator). Once I changed component values so smaller current was required, everything was great. All of this began right after I disconnected the device from the USB port while leaving the external power connected. The next morning, when I tried to turn things on, Scopy crashed.

Thanks, Ken

[image]<https://user-images.githubusercontent.com/8676912/97705115-35940900-1abc-11eb-82e7-18094fc52604.png>

connorka, 2020-10-30:
One more, I hope unrelated, issue. Whenever I plug in an M2K, my VPN disconnects and then reconnects on both laptops. I do not recall seeing that until just recently. I use a Cisco AnyConnect VPN through my university.

Ken

[image]<https://user-images.githubusercontent.com/8676912/97705115-35940900-1abc-11eb-82e7-18094fc52604.png>

adisuciu, 2020-10-30:
It could be because the M2K upon booting will create a new network interface. This reminds me, can you connect to the network interface using an IP address ? Instructions here:

https://wiki.analog.com/university/tools/m2k/scopy?s[]=remote#connecting_to_a_remote_device

Also my colleague is correct, the git tag suggests that you are still using the older version, however this might be due to clicking the report a bug button in the old version (different computer/etc).

If after trying to connect using IP, and using the latest version (1.2.0), it still fails, can you enable the debug messages so we can take a look ? 

If you are trying to attach files, you cannot do that through the email interface, you will have to access this link  - https://github.com/analogdevicesinc/scopy/issues/948

adisuciu, 2021-02-25:
This is not a release but this version should have a lot of fixes - would you mind giving it a try and see how it works?
This version has gain setting implemented for the voltmeter.
Make sure you tick delete the ini files in the setup process, or delete them manually.
https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/37944786/job/g4x6c5ep2csfs08o/artifacts
-Adrian

connorka, 2021-02-25:
Sure. Will take a day or two.
Ken

https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/37944786/job/g4x6c5ep2csfs08o/artifacts

adisuciu, 2021-02-25:
We didn't do anything specifically for your usecase, because we couldn't really reproduce, but hopefully some of the general fixes we added to the application will fix your issues. 
-Adrian

sja1440, 2021-08-01:
I just encountered the same symptoms on a laptop as described by the OP.  The crashes went away after I reset Scopy from Scopy->Preferences->Reset Scopy.

Here are some more details:

Laptop runs a fully updated Ubuntu 20.04.
Scopy is v1.3.0
ADALM2000 runs firmware v0.27

When I hit the problem I followed the instructions in adisuciu's post of 30/10/2020 on this thread.
I thought I might have a power issue so I attached an external power supply.
The problem still occurred so, as suggested by adisuciu, I set the Enable Session Logging flag in Scopy preferences and then hit the "Reset Scopy" button. 
At this point I made a mistake and exited Scopy. I restarted Scopy and connection was established without any issues but I saw no debugging information. Strange I thought but it seems that the setting of Enable Session Logging flag is volatile and so not carried through to the next Scopy start. 

I have done some tests - I am now unable to reproduce the problem even when not using an external power supply.  So it seems as if whatever caused my crash was not related to insufficient power but some setting within Scopy itself which was cleared by doing a Scopy reset.

I hope this is helpful to someone.