analogdevicesinc/scopy #949: Logic Analyzer improvements

adisuciu, 2020-11-02:
Logic analyzer:
- channel names could be 1/2/3 instead of DIO0/DIO1/DIO2 - this would make them look cleaner on the plot. The handle should have the text centered.
Oneshot maximum buffersize: MAX_BUFFER_SIZE*64 instead of MAX_BUFFER_SIZE - use kernel buffers similar to streaming
Streaming - sample rate max 10msps but buffer can be any size
Oneshot mode should be selected by default
Trigger delay should be visible for both modes

There is an plotting artifact when plotting lare buffersizes
[image]

When MAX_BUFFER_SIZE is reached in the pattern generator - show a warning
Show buffersize and sample rate on top of plot for both logic analyzer and pattern generator similar to oscilloscope.

[image]

DanielGuramulta, 2020-12-02:
Created a separate issue to track the bug described in the above list: #972

DanielGuramulta, 2021-01-11:
Implemented with: #971

tagoylo, 2021-04-07:
On "channel names could be 1/2/3 instead of DIO0/DIO1/DIO2 - this would make them look cleaner on the plot. The handle should have the text centered."
Is it only the channel names on the handle or is it also applicable on the list of channels on the menu? 

In logic analyzer, the name of the channel only changes the handle text and not the channel name list.
[image]

But in pattern generator, both changes.
[image]

DanielGuramulta, 2021-04-07:
> On "channel names could be 1/2/3 instead of DIO0/DIO1/DIO2 - this would make them look cleaner on the plot. The handle should have the text centered."
> Is it only the channel names on the handle or [...]

Things should be more consistent now with pr: https://github.com/analogdevicesinc/scopy/pull/1087

tagoylo, 2021-04-28:
> Streaming - sample rate max 10msps but buffer can be any size

does the sample max rate correspond to this parameter? Should this be only 10 msps?
[image]

adisuciu, 2021-04-28:
Yes this is fine, if you up the nr of samples past a threshold you will get a warning.

closed 2021-05-05 by adisuciu