analogdevicesinc/scopy #950: Show libm2k version in infopage

adisuciu, 2020-11-02:
[image]
Show libm2k version here

adisuciu, 2020-11-09:
https://github.com/analogdevicesinc/scopy/pull/959

adisuciu, 2020-11-27:
https://github.com/analogdevicesinc/scopy/pull/969

tagoylo, 2021-04-29:
libm2k version not shown in this info page
[image]

but it can be seen, in the About>build info
[image]

Is it intended to be shown on the info page?

adisuciu, 2021-05-05:
That's fine, the infopage shows device info (ADALM2000) which does not make use of libm2k

closed 2021-05-05 by adisuciu