analogdevicesinc/scopy #972: Logic tools plotting artefact

DanielGuramulta, 2020-12-02:
> * There is an plotting artifact when plotting lare buffersizes
> 
> [image]

DanielGuramulta, 2021-01-11:
Fixed with: #971

tagoylo, 2021-04-28:
Not sure if this is related, but this is what it looks like when sample rate is decreased. Are the lines considered the artifacts?
[image]

adisuciu, 2021-05-05:
Yes, they are artifacts, however, as opposed to the previous version, the plotting should now be correct. I'll keep investigating the artifact