analogdevicesinc/scopy #976: <Instrument>: <Short description of the bug>

3Ferro3, 2020-12-06:
OS Version: Windows 7 Version 6.1 (Build 7601: SP 1)
FW Version: 
SW Version: v1.2.0

Description of the bug:<Scopy non started and write problem on dump file>
Steps to reproduce:
Click on scopy icon
-

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/pc/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.
[image]
[image]

rgetz, 2020-12-23:
I don't think anyone can debug this with the information given.

What are you doing, when does it crash? (we need the files, not pictures of the files).

3Ferro3, 2020-12-24:
as required I attach the crash files. The crash happens when starting the application that does not open.
waiting for an answer. Kind Regards

Ferruccio Pacciani
[2e2384dc-452f-4153-8266-b524e318f71b.zip](https://github.com/analogdevicesinc/scopy/files/5739390/2e2384dc-452f-4153-8266-b524e318f71b.zip)

adisuciu, 2021-02-10:
We'll send over a test build in a few days. If you could test that, it would be great.

3Ferro3, 2021-02-11:
Thanks, as soon as the new version arrives I will try it immediately.

Starting Scopy nothing happens, the interface does not start, the dump files are written in the \ AppData \ Roaming \ ADI \ crashdmp \ folder.

Waiting for the new version to be tested

Kind regards

Ferruccio Pacciani

adisuciu, 2021-02-25:
This is not a release but this version should have a lot of fixes - would you mind giving it a try and see how it works?

Make sure you tick delete the ini files in the setup process, or delete them manually.

https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/37944786/job/g4x6c5ep2csfs08o/artifacts

-Adrian

adisuciu, 2021-05-05:
Closed due to inactivity - assuming fixed

closed 2021-05-05 by adisuciu