analogdevicesinc/scopy #977: Pattern Generator: application crashed when editing groups, cannot access DIO 0

lysynski, 2020-12-07:
OS Version: macOS Mojave (10.14)
FW Version: v0.26
SW Version: e02cebe

Description of the bug:application does not look as in wiki; started with creating a UART on DIO0, added DIO1, DIO2 as clock and 0, added DIO1 and 2 to a group, both became clks, tried to change DIO1 to 0 and application crashed. Relaunched application, can no longer edit DIO0.

Steps to reproduce:
occurs when laucnhed
-

The ini files might be useful to reproduce the error.
The ini file is located at: /Users/TTFG/.config/analog.com/Scopy.ini  

         - using MacOS, ini file not at this location

Please consider attaching it.

lysynski, 2020-12-07:
Additional information:

Cannot access settings for all DIO; settings are greyed out

adisuciu, 2020-12-07:
Yes, we're working on updating the wiki page for the logic analyzer.
Also you might have triggered this bug https://github.com/analogdevicesinc/scopy/pull/973 .. we're fixing it in the next version. 

Even though you are on macOS, the file should be at that location. If you are getting crashes on startup remove the scopy.ini file from that location and retry.

-Adrian

adisuciu, 2020-12-07:
https://wiki.analog.com/university/tools/m2k/scopy/logicanalyzer?rev=1607337491

lysynski, 2020-12-07:
Hi Adrian

So how do I reset the pattern generator so I can use it again?

The directory doesn’t exist; any other locations?

Thanks for the help.
Boris

adisuciu, 2020-12-07:
You should use the terminal to get to that location, because the .config folder is hidden. 
Regardless you can go to Preferences and click reset scopy. Close the application, and restart. The application should be reset

-Adrian

lysynski, 2020-12-07:
Sorry Adrian, I didn't have hidden files turned on - here's the ini file (remove .txt).
[Scopy.ini.txt](https://github.com/analogdevicesinc/scopy/files/5653910/Scopy.ini.txt)

lysynski, 2020-12-07:
Reset didn't restore the pattern generator; screenshot attached, you can see the pattern generator isn't available and within the settings are greyed out.
[image]
[image]

adisuciu, 2020-12-07:
Click the cogwheel and start enabling channels. The button will come up.
-Adrian

lysynski, 2020-12-07:
Have a look at the second screenshot; DIO0 is enabled, settings greyed out. It’s the same for all channels.

Boris

adisuciu, 2020-12-07:
You need to select the channels .. double click dio0 and a white outline will appear, the settings will be enabled.

lysynski, 2020-12-07:
Thanks, I see that now… I guess I did that without knowing it the first time.  “double click” the channel from the wiki led me to believe when I selected the channel.

Thanks again for the help.
Boris

adisuciu, 2020-12-07:
We have now updated the docs from the wiki. They should now resemble the new (and improved) logic analyzer. You can use that for reference.

closed 2020-12-07 by adisuciu