analogdevicesinc/scopy #989: ADALM2000 Refuses to connect

cacarm, 2021-01-13:
OS Version: Windows 10 Version 2004
FW Version: 
SW Version: v1.2.0

Description of the bug: I've tried what others have recommended: I've plugged it in before running scopy, I've had Scopy running while plugging it in, I've run as administrator, I've even restarted my laptop several times.
Steps to reproduce:
-
-

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/Carmen.LAPTOP-E75TJEIB/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.

adisuciu, 2021-01-13:
I don't understand. Does the ADALM2000 not show up ? Or does Scopy crash when connecting ?
-Adrian

cacarm, 2021-01-13:
Hi,
It just doesn’t show up. I saw in an introduction video that the icon
should show up on the top section after it’s plugged in, but it isn’t
there.

Thanks,
Carmen Ced

tfcollins, 2021-01-13:
Please make sure you have the drivers installed and can correctly talk to M2K. Follow this guide: https://wiki.analog.com/university/tools/m2k/users/quick_start

-Travis

tfcollins, 2021-01-21:
Closing for now.

-Travis

closed 2021-01-21 by tfcollins