analogdevicesinc/scopy #991: PatternGenerator: Interfering with DIO - locking unused channels

lysynski, 2021-01-21:
OS Version: macOS Mojave (10.14)
FW Version: v0.26
SW Version: e02cebe

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to DIO; set DIO 0, DIO1 to output
2. Go to PatternGenerator; create pattern on DIO 0, DIO 1
3. Run generator
4. Go to DIO
5. Observe DIO 0, DIO 1 are locked
6. Go to PatternGenerator; stop generator
7. Remove DIO 0; start generator
8. Go to DIO
9. Observe DIO 0 unlocked, DIO 1 locked
10. Change DIO 0 value, observe output; does not change, still locked, output from PatternGenerator

**Expected behavior**
I expect DIO 0 to behave normaly under DIO control.

**Additional context**

[Scopy.ini.txt](https://github.com/analogdevicesinc/scopy/files/5851823/Scopy.ini.txt)

adisuciu, 2021-01-22:
Thanks, we're looking into it
-Adrian

adisuciu, 2021-01-22:
Thanks, we're looking into it. Did you run the DIO instrument - outputs from DIO are enabled only when DIO is running (using the green "run" button top right)
-Adrian

lysynski, 2021-01-22:
Correct, PatternGenrator and  DIO are running.

Boris

lysynski, 2021-02-01:
Hi Adrian,
Do you have an update for this, are you making progress?  Thanks.

Boris

adisuciu, 2021-02-02:
Hey, not yet. It's on our TODO list for next release though, so it will be handled before the next release.

lysynski, 2021-02-02:
It’s good you found it.  When’s the next release due?

adisuciu, 2021-02-03:
We're hoping to wrap things up this month ..

lysynski, 2021-03-02:
Hi Adrian,

I went to get the latest release which shows Feb 8, 2021, when I install it’s the same release as before v1.2.0 - e02cebe.  Is there a different location for it?  I’m going to:

https://wiki.analog.com/university/tools/m2k/scopy <https://wiki.analog.com/university/tools/m2k/scopy>

Thanks

tagoylo, 2021-04-30:
Verified fix on v1.3 RC2. 
macOS version: Big Sur 11.2.3.

lysynski, 2021-04-30:
Hi Trecia

Thanks for the information; do you have a version macOS 10 i.e. Mojave?

Boris

lysynski, 2021-04-30:
Sorry, I notice I left off the 14;  macOS 10.14

Thanks

adisuciu, 2021-05-04:
https://github.com/analogdevicesinc/scopy/releases/tag/v1.3.0-rc2 - @lysynski  you can test this here

lysynski, 2021-05-04:
Hi Adrian,

Thanks for the link but that’s for macOS 10.15; I’m looking for 10.14.

Boris

adisuciu, 2021-07-05:
We created a new release. Your issue should be fixed

closed 2021-07-05 by adisuciu