analogdevicesinc/scopy #996: Voltmeter reading is out of spec and is different to Oscilloscope

mickatclearbox, 2021-02-08:
OS Version: Windows 10 Version 2004
FW Version: v0.25
SW Version: v1.2.0

Description of the bug: Voltmeter reading near 
Steps to reproduce:
Connecting up a 1.4 volt D cell battery 
Measured with Scopy voltmeter  1.236 V
Measured with Scopy voltmeter in reverse polarity -1.353 V  
Measured with Scopy Oscilloscope 1.322 V
Measured with Scopy Oscilloscope in reverse polarity -1.322 V
For comparison, measure using calibrated Keithley DMM6500 1.31537 V

Looks like some problem with the Scopy Voltmeter?  would expect both reverse polarity measurement to be the sames as the forward x -1.  Also would expect the Scopy Oscilloscope and the Scopy voltmeter to closely match.   

-

The ini files might be useful to reproduce the error.
[Scopy.zip](https://github.com/analogdevicesinc/scopy/files/5940928/Scopy.zip)

Please consider attaching it.

mickatclearbox, 2021-02-08:
Same wrong behavior with FW Version: v0.26

adisuciu, 2021-02-10:
We're looking into it ..

mickatclearbox, 2021-02-11:
Great, let me know if you need more info, testing or something else.

adisuciu, 2021-02-25:
This is not a release but this version should have a lot of fixes - would you mind giving it a try and see how it works?
This version has gain setting implemented for the voltmeter.
Make sure you tick delete the ini files in the setup process, or delete them manually.
https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/37944786/job/g4x6c5ep2csfs08o/artifacts
-Adrian

mickatclearbox, 2021-04-21:
I updated Scopy to v1.3.0-rc1 441Fd19 and M2K FW to V0.27.

This issue still exist and is not fixed.  There a 70mV difference between the voltmeter and the oscilloscope readings.  The oscilloscope reading is close about 10mv differences to my Keithley calibrated 6.5 digit DMM6500 desktop meter, so this looks correct.  As the Voltmeter and Oscilloscope use the same hardware channel they should be the same. But the Scopy Voltmeter is still  reading incorrectly, about 70mv lower then Scopy oscilloscope reading.  attached pic of version and cal details of my M2K and Scopy

[image]

adisuciu, 2021-05-05:
Can you update to rc2 - there is a fix in there for the multimeter .. https://github.com/analogdevicesinc/scopy/releases/tag/v1.3.0-rc2

mickatclearbox, 2021-05-07:
v1.3.0-rc2 is not finding ADALM2000. Tried rebooting, reconnecting M2K etc. . . nothing help. I uninstalled rc2 and re install with tick on driver install, reboot, then connect M2K and still Scopy did not find M2K.    Can see M2k in USB drive so its connected.

adisuciu, 2021-05-07:
Hmm, that's weird .. Just to troubleshoot, can you try connecting via the remote interface ?  https://wiki.analog.com/university/tools/m2k/scopy#connecting_to_a_remote_device .. can ypu see the driver in the device manager like in the picture ? https://wiki.analog.com/university/tools/pluto/drivers/windows ?
-Adrian

adisuciu, 2021-10-20:
Closed due to inactivity

closed 2021-10-20 by adisuciu