analogdevicesinc/scopy #997: Osciloscope: switching to AC crashes the instrument

Rmano, 2021-02-09:
**Environment:**
 - OS: Linux
 - FW 0.26
 - SW 1.2.0
 - instrument attached via USB

**Describe the bug and To Reproduce**
Switch the oscilloscope on, Run, select "Software ac coupling"
The Scopy application crashes instantly.

**Additional context**

When crashing, the message is the following:

    SYS: Feb  9 10:39:53 samsung-romano kernel: [ 6616.263811] traps: repeat44[22401] trap invalid opcode    ip:7f0ddd7f3a3b sp:7f0d217f95c8 error:0 in libgmp.so.10.4.0[7f0ddd7d4000+59000]
    SYS: Feb  9 10:39:53 samsung-romano systemd[3946]: flatpak-org.adi.Scopy-22218.scope: Succeeded.

**ini file** 

Which is the `.ini` file relevant for Linux? Will post if you tell me ;-)

adisuciu, 2021-02-10:
That's weird. What system are you running ? Are you running flatpak ?
The inifile is in .config/ADI/Scopy.ini

I couldn't reproduce any of your issues (neither this/ nor the one in https://github.com/analogdevicesinc/scopy/issues/998 ) but these seem related to eachother .

We will finish merging some fixes, and we can send you another version of Scopy to test. In a week or so.
-Adrian

Rmano, 2021-02-10:
Hi --- running Ubuntu 20.04 on a Samsung Laptop when having the issue, i7 with Intel graphics. I am not at home now but if you want I can check it. 

Yes, I am running the flatpack version (is there another one to test?)
I can confirm that a colleague of mine here at the university (we are testing the ADALM2000 together) have the same problem with the AC/DC switching thing.

Thanks, looking forward to check more. Very nice application, by the way!

adisuciu, 2021-02-10:
Just to be sure, send us the Preferences.ini as well . We're gonna have a new flatpak ready in a few days. I'll send over the link when it's done.

Rmano, 2021-02-10:
Here it goes: 
[DotConfigADI.zip](https://github.com/analogdevicesinc/scopy/files/5960759/DotConfigADI.zip)

Thanks!

adisuciu, 2021-03-17:
We may have fixed this here - https://github.com/analogdevicesinc/scopy/issues/1050
Can you give it a try ? https://github.com/analogdevicesinc/scopy/suites/2244955973/artifacts/47401710

Rmano, 2021-03-17:
Yep! I will try to check it tomorrow... thanks!

Rmano, 2021-03-18:
Hi! Tested, almost fully good news. 

1. The crashes are gone --- thanks a lot!
2. Network Analyzer is working (great!)
3. Spectrum Analyzer is not crashing anymore but it doesn't work, there is no output visible. 

During the installation of one runtime I had the following error:

```
3. [✗] org.freedesktop.Platform.openh264                                    2.0             i          flathub                 1.2 MB / 1.5 MB
Warning: Error deploying: While trying to apply extra data: apply_extra script failed, exit status 256
```

which I do not know if it's related.

Rmano, 2021-03-18:
@adisuciu  ^^^^ forgot the ping, sorry

adisuciu, 2021-03-18:
Can you reset the .ini file by deleting it ? Before deletion, can you send it over to us to check out if there's anything wrong there ? https://wiki.analog.com/university/tools/m2k/scopy-troubleshooting#ini_files

This issue - https://github.com/analogdevicesinc/scopy/issues/966#issuecomment-801998579 -  behaves similarly, no data shown on the spectrum analyzer, but things seemd to work after resetting.

Rmano, 2021-03-18:
Ok --- I have a bit of info for you. These are the INI files before removal:
[before.zip](https://github.com/analogdevicesinc/scopy/files/6165672/before.zip)

So, I removed the files, and now it works but it has a couple glitches. I have generated this signal and connected it to channel 1 of the scope:
[image]

At first, the spectrum analyzer seems not to work. Then I noticed that the displayed range was far off --- adjusting it resulted in a nice spectrum. BUT:

- as soon as you change anything (range of frequency, center, whatever), the spectrum calculation stops, without reflecting it in the button run/stop (see, I changed range here):

[image]

This is solved by choosing a bigger resolution BW --- you choose a too-small one, the display freezes without any message/error status. But yes, basically once you know it, it works! Thanks.

[image]

adisuciu, 2021-03-20:
Can you tell exactly what operation caused the spectrum range to change ? Something like steps for reproduction the issue ?

Rmano, 2021-03-21:
I simply changed (upped) the "stop" value while the display was running. I will double check later!

adisuciu, 2021-03-21:
My point is that removing the ini file should reset the spectrum range, so I was wondering what operation caused the range to be off (as you were describing it). If you changed it manually, then there is no bug. I was trying to see if it was a similar scenario to what we have here - https://github.com/analogdevicesinc/scopy/issues/1055

closed 2021-07-05 by adisuciu