analogdevicesinc/scopy #1010: Scopy fails to start

morganjohansson, 2021-02-21:
**Environment:**
 - OS: Windows 10.0.19042
 - FW 0.26 on the ADALM2000
 - SW Scopy 1.2.0

**Describe the bug**
Firmware and drivers installs and all checks are passed
No errors when installing Scopy.
When starting Scopy it runs for a couple of seconds and terminates silently.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/ali/AppData/Roaming/ADI/Scopy.ini

There is no Scopy.ini to find anywhere on the disk.
There is no folder C:/Users/morgan/AppData the closest I have is 
C:\Users\morgan\Roaming\Intel\Wireless

adisuciu, 2021-02-25:
This is not a release but this version should have a lot of fixes - would you mind giving it a try and see how it works?
Make sure you tick delete the ini files in the setup process, or delete them manually.
https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/37944786/job/g4x6c5ep2csfs08o/artifacts
-Adrian

morganjohansson, 2021-02-25:
Hi Adrian
Thanks a lot, this version worked, started connected and calibrated.
Now I'll start to try the different tools.

/Morgan Johansson

closed 2021-02-25 by morganjohansson