analogdevicesinc/scopy #1012: <Instrument>: <Short description of the bug>

EverDomin935, 2021-02-23:
OS Version: Windows 10 Version 2009
FW Version: 
SW Version: v1.2.0

Description of the bug:<description>
Steps to reproduce:
-
-start scopy
iclick on connect
-it crashes immediately

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/wintr/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.
[DMMSpinBoxTimer12]
fine_mode=false

[Debug]
detached=false

[Digital%20IO]
detached=false

[Logic%20Analyzer]
detached=false

[Network%20Analyzer]
detached=false

[NetworkAnalyzerSpinBoxAmplitude30]
fine_mode=false

[NetworkAnalyzerSpinBoxMax.Magnitude32]
fine_mode=false

[NetworkAnalyzerSpinBoxMax.Phase34]
fine_mode=false

[NetworkAnalyzerSpinBoxMin.Magnitude33]
fine_mode=false

[NetworkAnalyzerSpinBoxMin.Phase35]
fine_mode=false

[NetworkAnalyzerSpinBoxOffset31]
fine_mode=false

[NetworkAnalyzerSpinBoxSamplescount27]
fine_mode=false

[NetworkAnalyzerSpinBoxSamps]
decade28\fine_mode=false

[NetworkAnalyzerSpinBoxSettlingtime36]
fine_mode=false

[NetworkAnalyzerSpinBoxSettlingtime37]
fine_mode=false

[NetworkAnalyzerSpinBoxStep29]
fine_mode=false

[Oscilloscope]
detached=false

[OscilloscopeSpinBoxPosition11]
fine_mode=false

[OscilloscopeSpinBoxPosition9]
fine_mode=false
[... 575 lines cut ...]
markers\9\visible=false
markers\9\type=-1
markers\9\freq=0
markers\10\chId=1
markers\10\mkId=4
markers\10\en=false
markers\10\visible=false
markers\10\type=-1
markers\10\freq=0
logScale=false
notes=

[toolMenu]
pos\1\idx=0
pos\2\idx=1
pos\3\idx=2
pos\4\idx=3
pos\5\idx=4
pos\6\idx=5
pos\7\idx=6
pos\8\idx=7
pos\9\idx=8
pos\10\idx=9
pos\11\idx=10
pos\size=11

adisuciu, 2021-02-25:
This is not a release but this version should have a lot of fixes - would you mind giving it a try and see how it works?
Make sure you tick delete the ini files in the setup process, or delete them manually.
https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/37944786/job/g4x6c5ep2csfs08o/artifacts
-Adrian

adisuciu, 2021-04-22:
Closing this due to inactivity - reopen if you encounter any more issues ..

closed 2021-04-22 by adisuciu