analogdevicesinc/scopy #1018: ADALM2000: Device not recognized by Linux or Scopy (Linux Mint 20.1)

nickr67, 2021-02-26:
OS Version: KDE Flatpak runtime
FW Version: 
SW Version: e02cebe

Description of the bug:<description>
Steps to reproduce:
- I followed the install routine here https://wiki.analog.com/university/tools/m2k/scopy#connecting_to_a_usb_device
- I connected the device and nothing. Not even a peep from Linux.
- There are no ini files at the given location.

The ini files might be useful to reproduce the error.
The ini file is located at: /home/nicky/.var/app/org.adi.Scopy/config/ADI/Scopy.ini
Please consider attaching it.

adisuciu, 2021-02-26:
https://wiki.analog.com/university/tools/pluto/drivers/linux?s[]=drivers&s[]=pluto
Did you install the udev rules ? 

```
In order to access some USB functions without root privileges, it's recommended to install the PlutoSDR or ADALM2000 udev rules. Simply download 53-adi-plutosdr-usb.rules or 53-adi-m2k-usb.rules and copy into into the /etc/udev/rules.d/ folder. You can also use this plutosdr-m2k-udev.deb package to install both the PlutoSDR and the ADALM2000 udev rules. To install the debian package, use one of the following: sudo apt-get install ./plutosdr-m2k-udev.deb or sudo dpkg -i plutosdr-m2k-udev.deb.
Afterwards reload rules or restart udev, using either udevadm control --reload-rules or sudo service udev restart
```

nickr67, 2021-02-26:
Mr. Suciu,

That may be the case, however, I think my ADALM was damaged during a thunderstorm last night. My Windows installation does not recognize it either.
Thanks for quick response though.

I tried that solution, but I was unable to locate the file location.
Thanks.

- Nicolas Rodriguez

adisuciu, 2021-02-26:
Windows installation needs a driver as well https://wiki.analog.com/university/tools/pluto/drivers/windows?s[]=driver&s[]=windows&s[]=pluto

in linux dmesg should let you know that you connected a device. 
Alternatively you can try connecting via the network interface in linux(or Windows) which does not require a driver. Simply add a new device and specify the default ip: 192.168.2.1 - https://wiki.analog.com/university/tools/m2k/scopy#connecting_to_a_remote_device

The ADALM2000 when working correctly should have one blue LED on and the  other LED blink in a "heartbeat"  kind of way. If the heartbeat pattern is not present, it is possible it is damaged.
-Adrian

nickr67, 2021-02-26:
Mr. Suciu,

Thanks again.
The Ready LED is not flashing. Only the LED1 is turned on. Default IP connection did not work.

I think the thunderstorm killed it. RIP.

adisuciu, 2021-02-26:
I'm sorry to hear this ... 
Just in case it wasn't the thunderstorm, and the firmware got corrupted by other means, there is a guide here to try to resurrect it https://wiki.analog.com/university/tools/m2k/common/firmware but it's kinda long shot.

I'm closing this. Reopen it if you have any other issues.

closed 2021-02-26 by adisuciu