analogdevicesinc/scopy #1031: <SCOPY>: <No response>

avatarof, 2021-03-08:
**Environment:**
 - OS: win10
 - FW Version  V0.26
 - SW Version or git tag V1.2.0-d2a467f

**Describe the bug**
Application No response.
[image]

**To Reproduce**
Any operation can randomly trigger this bug. 

**Expected behavior**
 Response.

**Additional context**

[Scopy.txt](https://github.com/analogdevicesinc/scopy/files/6099051/Scopy.txt)

rgetz, 2021-03-08:
To reproduce, we need more information that “any operation”.  What exactly did you do?

adisuciu, 2021-03-09:
Thanks for the video. It was very helpful. 

It looks like this is a regression with upgrading to libusb v1.0.24 . We're testing things out. We can either downgrade to v1.0.21(which is stable) or see if it's a problem on libusb side or on our side.

In order to confirm, you can try connecting to the ADALM2000 using the remote context. Add the default ADALM2000 ip to Scopy - 192.168.2.1 and connect to that device - https://wiki.analog.com/university/tools/m2k/scopy?s[]=connecting&s[]=remote#connecting_to_a_remote_device. 

Things should work pretty much the same without crashing. You can give it a try without reinstalling anything to confirm.

-Adrian

avatarof, 2021-03-10:
> Thanks for the video. It was very helpful.
>
> It looks like this is a regression with upgrading to libusb v1.0.24 . We're testing things out. We can either downgrade to v1.0.21(which is stable) or see [...]

Tried.

Still appeared no response
[image]
Not tried for a very long time, so cannot say if the [no response] freq. go down or not.

adisuciu, 2021-03-10:
That is weird .. we couldn't reproduce the hang when connecting via IP. I attached a small video. 

https://user-images.githubusercontent.com/8676912/110607371-cc23b580-8193-11eb-9e8f-ae853da2ace1.mp4

We'll keep investigating.

adisuciu, 2022-03-10:
Reopen if still an issue

closed 2022-03-10 by adisuciu