analogdevicesinc/scopy #1032: Signal Generator: Wrong preview scale

avatarof, 2021-03-08:
**Environment:**
OS: win10
FW Version V0.26
SW Version or git tag V1.2.0-d2a467f

**Describe the bug**
Wrong preview scale.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Signal Generator
2. sine wave
3. amp.=50mV p2p
4. see the plot
**Expected behavior**
Right amp. scale

**Additional context**

[Scopy .txt](https://github.com/analogdevicesinc/scopy/files/6099085/Scopy.txt)

kimpaller, 2021-04-29:
Issue is still replicated on the following environment:

OS: Windows 10
FW Version: https://github.com/analogdevicesinc/m2k-fw/releases/tag/v0.27
SW Version or git tag: [v1.3.0-rc2](https://github.com/analogdevicesinc/scopy/releases/tag/v1.3.0-rc2)

[image]

tagoylo, 2022-04-29:
Issue not reproducible anymore in v1.4.0 rc2. Closing this issue.

closed 2022-04-29 by tagoylo