analogdevicesinc/scopy #1049: ADALM2000: Scopy build errors

jpnielsen, 2021-03-15:
Debian 11 testing

Can't build scopy on Debian 11 (Bullseye)

I tried builing Scopy from git on Debian 11 with QT5 and Qt6 - no success, see below

The flatpak image is giving me problems too - i will describe that in another "Issue".

####################

```
Attempts at building ADI scopy on Debian 11, with a fresh Qt from qt.io

sudo apt install build-essential git

cd ~ 
mkdir git_build
cd git_build
git clone https://github.com/analogdevicesinc/scopy.git

Installed "Qt 6.0.2 for GCC for desktop application development" from qt.io to /home/jpn/Qt6

wget http://archive.ubuntu.com/ubuntu/pool/universe/libz/libzip/libzip5_1.5.1-0ubuntu1_amd64.deb
dpkg -i libzip5_1.5.1-0ubuntu1_amd64.deb

scopy/CI/appveyor/install_ubuntu_deps.sh /home/Adi/Qt/5.15.0/gcc_64

There is no "install_ubuntu_deps.sh", so I changed to :

scopy/CI/appveyor/install_ubuntu_20_deps.sh ~/Qt6/6.0.2

...scrolling text for a while...
 
    qwt_plot_canvas.cpp: In member function ‘virtual void QwtPlotCanvas::drawBorder(QPainter*)’:
    qwt_plot_canvas.cpp:1023:13: error: ‘class QStyleOptionFrame’ has no member named ‘init’
    1023 |         opt.init(this);
        |             ^~~~
    make[1]: *** [Makefile:2068: obj/qwt_plot_canvas.o] Error 1
    make[1]: Leaving directory '/home/jpn/qwt/src'
    make: *** [Makefile:50: sub-src-make_first-ordered] Error 2

Installed "Qt 5.12.2 for GCC for desktop application development" from qt.io to ~/Qt5

cd git_build
scopy/CI/appveyor/install_ubuntu_20_deps.sh ~/Qt5/5.12.2

rm -rf ~/glog ~/gr-iio ~/gr-m2k ~/gr-scopy ~/libad9361 ~/libiio ~/libm2k ~/libsigrokdecode ~/qwt ~/qwtpolar 

the build script will not run if the destination dirs are there and not empty.

please keep the files below ~/git_build, where i ran the script !!
remove line 28 in https://github.com/analogdevicesinc/scopy/blob/master/CI/appveyor/install_ubuntu_20_deps.sh

cmake -DCMAKE_PREFIX_PATH=~/Qt5/5.15.2/gcc_64/lib/cmake ../ 
make -j4

...scrolling text for a while...

 Linking CXX executable scopy
/usr/bin/cmake -E cmake_link_script CMakeFiles/scopy.dir/link.txt --verbose=1
/usr/bin/c++ -O2 -g -DNDEBUG CMakeFiles/scopy.dir/scopy_autogen/mocs_compilation.cpp.o CMakeFiles/scopy.dir/src/ComboBoxLineEdit.cpp.o CMakeFiles/scopy.dir/src/ConstellationDisplayPlot.cc.o CMakeFiles/scopy.dir/src/DisplayPlot.cc.o CMakeFiles/scopy.dir/src/FftDisplayPlot.cc.o CMakeFiles/scopy.dir/src/HistogramDisplayPlot.cc.o CMakeFiles/scopy.dir/src/TimeDomainDisplayPlot.cc.o CMakeFiles/scopy.dir/src/adc_sample_conv.cpp.o CMakeFiles/scopy.dir/src/animationmanager.cpp.o CMakeFiles/scopy.dir/src/apiObject.cpp.o CMakeFiles/scopy.dir/src/apiobjectmanager.cpp.o CMakeFiles/scopy.dir/src/application_restarter.cpp.o CMakeFiles/scopy.dir/src/autoScaler.cpp.o CMakeFiles/scopy.dir/src/average.cpp.o CMakeFiles/scopy.dir/src/basemenu.cpp.o CMakeFiles/scopy.dir/src/basemenuitem.cpp.o CMakeFiles/scopy.dir/src/bitfieldwidget.cpp.o CMakeFiles/scopy.dir/src/buffer_previewer.cpp.o CMakeFiles/scopy.dir/src/calibration.cpp.o CMakeFiles/scopy.dir/src/calibration_api.cpp.o CMakeFiles/scopy.dir/src/cancel_dc_offset_block.cpp.o CMakeFiles/scopy.dir/src/channel_widget.cpp.o CMakeFiles/scopy.dir/src/checkbox_delegate.cpp.o CMakeFiles/scopy.dir/src/coloredQWidget.cpp.o CMakeFiles/scopy.dir/src/completion_circle.cpp.o CMakeFiles/scopy.dir/src/connectDialog.cpp.o CMakeFiles/scopy.dir/src/cursor_readouts.cpp.o CMakeFiles/scopy.dir/src/customFifo.cpp.o CMakeFiles/scopy.dir/src/customPushButton.cpp.o CMakeFiles/scopy.dir/src/customScale.cpp.o CMakeFiles/scopy.dir/src/customSwitch.cpp.o CMakeFiles/scopy.dir/src/customanimation.cpp.o CMakeFiles/scopy.dir/src/customplotpositionbutton.cpp.o CMakeFiles/scopy.dir/src/db_click_buttons.cpp.o CMakeFiles/scopy.dir/src/dbgraph.cpp.o CMakeFiles/scopy.dir/src/debug.cpp.o CMakeFiles/scopy.dir/src/debugger.cpp.o CMakeFiles/scopy.dir/src/detachdragzone.cpp.o CMakeFiles/scopy.dir/src/detachedWindow.cpp.o CMakeFiles/scopy.dir/src/detachedwindowsmanager.cpp.o CMakeFiles/scopy.dir/src/device_widget.cpp.o CMakeFiles/scopy.dir/src/digitalchannel_manager.cpp.o CMakeFiles/scopy.dir/src/digitalio.cpp.o CMakeFiles/scopy.dir/src/digitalio_api.cpp.o CMakeFiles/scopy.dir/src/dmm.cpp.o CMakeFiles/scopy.dir/src/dmm_api.cpp.o CMakeFiles/scopy.dir/src/dragzone.cpp.o CMakeFiles/scopy.dir/src/drop_untill_tag_arives.cpp.o CMakeFiles/scopy.dir/src/dropdown_switch_list.cpp.o CMakeFiles/scopy.dir/src/dynamicWidget.cpp.o CMakeFiles/scopy.dir/src/extendingplotzoomer.cpp.o CMakeFiles/scopy.dir/src/external_script_api.cpp.o CMakeFiles/scopy.dir/src/fft_block.cpp.o CMakeFiles/scopy.dir/src/filemanager.cpp.o CMakeFiles/scopy.dir/src/filter.cpp.o CMakeFiles/scopy.dir/src/frequency_compensation_filter_impl.cc.o CMakeFiles/scopy.dir/src/graticule.cpp.o CMakeFiles/scopy.dir/src/handles_area.cpp.o CMakeFiles/scopy.dir/src/handlesareaextension.cpp.o CMakeFiles/scopy.dir/src/hardware_trigger.cpp.o CMakeFiles/scopy.dir/src/histogram_sink_f_impl.cc.o CMakeFiles/scopy.dir/src/homepage_controls.cpp.o CMakeFiles/scopy.dir/src/hw_dac.cpp.o CMakeFiles/scopy.dir/src/iio_manager.cpp.o CMakeFiles/scopy.dir/src/info_page.cpp.o CMakeFiles/scopy.dir/src/info_widget.cpp.o CMakeFiles/scopy.dir/src/instrumentnotes.cpp.o CMakeFiles/scopy.dir/src/jsfileio.cpp.o CMakeFiles/scopy.dir/src/lcdNumber.cpp.o CMakeFiles/scopy.dir/src/limitedplotzoomer.cpp.o CMakeFiles/scopy.dir/src/logging_categories.cpp.o CMakeFiles/scopy.dir/src/logic_analyzer_sink_impl.cpp.o CMakeFiles/scopy.dir/src/logic_tool.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/annotation.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/annotationcurve.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/annotationdecoder.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/binding/binding.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/binding/decoder.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/decoder.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/genericlogicplotcurve.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/logic_analyzer.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/logicanalyzer_api.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/logicdatacurve.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/logicgroupitem.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/prop/bool.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/prop/double.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/prop/enum.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/prop/int.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/prop/property.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/prop/string.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/row.cpp.o CMakeFiles/scopy.dir/src/logicanalyzer/rowdata.cpp.o CMakeFiles/scopy.dir/src/main.cpp.o CMakeFiles/scopy.dir/src/manual_calibration_api.cpp.o CMakeFiles/scopy.dir/src/manualcalibration.cpp.o CMakeFiles/scopy.dir/src/marker_controller.cpp.o CMakeFiles/scopy.dir/src/marker_table.cpp.o CMakeFiles/scopy.dir/src/math.cpp.o CMakeFiles/scopy.dir/src/measure.cpp.o CMakeFiles/scopy.dir/src/measure_settings.cpp.o CMakeFiles/scopy.dir/src/measurement_gui.cpp.o CMakeFiles/scopy.dir/src/menu_anim.cpp.o CMakeFiles/scopy.dir/src/mixed_signal_sink_impl.cpp.o CMakeFiles/scopy.dir/src/network_analyzer.cpp.o CMakeFiles/scopy.dir/src/network_analyzer_api.cpp.o CMakeFiles/scopy.dir/src/networkanalyzerbufferviewer.cpp.o CMakeFiles/scopy.dir/src/nyquistGraph.cpp.o CMakeFiles/scopy.dir/src/nyquistplotzoomer.cpp.o CMakeFiles/scopy.dir/src/osc_adc.cpp.o CMakeFiles/scopy.dir/src/osc_adjuster.cpp.o CMakeFiles/scopy.dir/src/osc_capture_params.cpp.o CMakeFiles/scopy.dir/src/osc_custom_scroll.cpp.o CMakeFiles/scopy.dir/src/osc_export_settings.cpp.o CMakeFiles/scopy.dir/src/osc_import_settings.cpp.o CMakeFiles/scopy.dir/src/osc_scale_engine.cpp.o CMakeFiles/scopy.dir/src/osc_scale_zoomer.cpp.o CMakeFiles/scopy.dir/src/oscilloscope.cpp.o CMakeFiles/scopy.dir/src/oscilloscope_api.cpp.o CMakeFiles/scopy.dir/src/oscilloscope_plot.cpp.o CMakeFiles/scopy.dir/src/paintersaverestore.cpp.o CMakeFiles/scopy.dir/src/patterngenerator/pattern_generator.cpp.o CMakeFiles/scopy.dir/src/patterngenerator/pattern_generator_api.cpp.o CMakeFiles/scopy.dir/src/patterngenerator/patterns/patterns.cpp.o CMakeFiles/scopy.dir/src/phonehome.cpp.o CMakeFiles/scopy.dir/src/plot_line_handle.cpp.o CMakeFiles/scopy.dir/src/plot_utils.cpp.o CMakeFiles/scopy.dir/src/plotpickerwrapper.cpp.o CMakeFiles/scopy.dir/src/power_controller.cpp.o CMakeFiles/scopy.dir/src/power_controller_api.cpp.o CMakeFiles/scopy.dir/src/preferences.cpp.o CMakeFiles/scopy.dir/src/printableplot.cpp.o CMakeFiles/s [... cut ...]
/usr/bin/ld: /usr/local/lib/libsigrokdecode.so: undefined reference to `PyList_New'
/usr/bin/ld: /usr/local/lib/libsigrokdecode.so: undefined reference to `PyTuple_Size'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/scopy.dir/build.make:3387: scopy] Error 1
make[2]: Leaving directory '/home/jpn/git_build/scopy/build'
make[1]: *** [CMakeFiles/Makefile2:99: CMakeFiles/scopy.dir/all] Error 2
make[1]: Leaving directory '/home/jpn/git_build/scopy/build'
make: *** [Makefile:152: all] Error 2
```

adisuciu, 2021-03-16:
It looks like python3 does not correctly link to libsigrokdecode. Are you using python 3.9 ? I'm not sure if the sigrok version provided by the installer has python 3.9 support as this was added only in august 2020.  
You can also try using libsigrokdecode from the apt repository. 

Is there any way I can easily test debian 11 in a VM. I browsed around the debian site, but noticed that it was not released yet and couldn't find a nightly ISO installer.
-Adrian

jpnielsen, 2021-03-16:
Thank you for looking into this.

I can change my linux distributuin, if necessary - what do you recommend to be able to build the latest from ADI?

answers below.

best regards
Jens Peter Nielsen

$ python3 --version
Python 3.9.2

I will try to find a newer version of libsigrokdecode if possible.

I get my debian testing from here:

https://cdimage.debian.org/cdimage/unofficial/non-free/cd-including-firmware/weekly-builds/amd64/iso-cd/firmware-testing-amd64-netinst.iso

After installation, I have changed from testing aka bullseye to "sid" - to get the latest versions available from the Debian people.

I do this by changing /etc/apt/sources.list to read:

```
deb http://deb.debian.org/debian sid main contrib non-free
deb-src http://deb.debian.org/debian sid main contrib non-free

```
and then "apt update && apt dist-upgrade"

adisuciu, 2021-03-16:
If you want stability I'd just use ubuntu 20.04. I'll try the latest debian but it's probably a matter of installing the correct libraries (installing libsigrokdecode from apt as opposed from source, or installing the correct version of libsigrokdecode which has python3.9 support)
-Adrian

jpnielsen, 2021-03-16:
Thank you for the advice. I'll close the issue now, now that the flatpak runs, it's not very important to me to be able to build Scopy from git.

closed 2021-03-16 by jpnielsen

jpnielsen, 2021-03-19:
I can confim, that scopy builds on Ubuntu 20.10 (Groovy Gorilla)

```
sudo apt install build-essential qt5-qmake qt5-dev qt5-default libqt5svg5-dev qttools5-dev qtdeclarative5-dev libqt5qml5

export MAKEFLAGS="-j4"
git clone https://github.com/analogdevicesinc/scopy.git

in scopy/CI/appveyor/install_ubuntu_20_deps.sh 
changed multiple occurences "cd ~" to "cd {WORKDIR}"

scopy/CI/appveyor/install_ubuntu_20_deps.sh
cd scopy/ 
mkdir build 
cd build
cmake ../
make
```